travisgoodspeed/md380tools #775: updatedb failed

cyurek, 2017-06-03:
all went well with the flash to my md380 except when I tried to run the updatedb here is the following error.

pi@raspberrypi3:~/md380tools $ make updatedb flashdb
"make" -C db update
make[1]: Entering directory '/home/pi/md380tools/db'
"make" clean
make[2]: Entering directory '/home/pi/md380tools/db'
rm -f repeaters.csv users.json repeaters.json special_IDs.csv *.tmp stripped.csv
make[2]: Leaving directory '/home/pi/md380tools/db'
"make" stripped.csv
make[2]: Entering directory '/home/pi/md380tools/db'
python2 get_special_IDs.py
Fetching list of special IDs from BM master servers.

0: ID->2021 Country->GR Address->master.dv4gr.eu
List with special IDs not found!

1: ID->2042 Country->NL Address->213.222.29.197
List with special IDs not found!

2: ID->2048 Country->NL Address->hoseline.pd0mz.nl
Could not talk to master server in NL!

3: ID->2049 Country->NL Address->83.86.45.116
List with special IDs not found!

4: ID->2082 Country->FR Address->217.182.129.131
List with special IDs not found!

5: ID->2141 Country->ES Address->84.232.5.113
List with special IDs not found!

6: ID->2161 Country->HU Address->brandmeister.sharkrf.com
List with special IDs not found!

7: ID->2162 Country->HU Address->master.brandmeister.hu
List with special IDs not found!

8: ID->2221 Country->IT Address->brandmeister.digitalham.it
List with special IDs not found!

9: ID->2222 Country->IT Address->brandmeister2.digitalham.it
List with special IDs not found!

10: ID->2229 Country->IT Address->bmhose.digitalham.it
List with special IDs not found!

11: ID->2261 Country->RO Address->77.81.108.136
List with special IDs not found!

12: ID->2281 Country->CH Address->128.65.196.21
List with special IDs not found!

13: ID->2302 Country->CZ Address->80.250.21.206
List with special IDs not found!

14: ID->2342 Country->GB Address->91.121.101.163
List with special IDs not found!

15: ID->2381 Country->DK Address->85.90.246.225
List with special IDs not found!

16: ID->2401 Country->SE Address->83.233.234.102
List with special IDs not found!

17: ID->2421 Country->NO Address->bm.hamlabs.no
List with special IDs not found!

18: ID->2441 Country->FI Address->brandmeister.oh2ch.org
List with special IDs not found!

19: ID->2502 Country->RU Address->brandmeister.r2aee.net
Could not talk to master server in RU!

20: ID->2503 Country->RU Address->164.215.71.108
List with special IDs not found!

21: ID->2508 Country->RU Address->hose.r2aee.net
List with special IDs not found!

22: ID->2602 Country->PL Address->195.26.76.59
List with special IDs not found!

23: ID->2622 Country->DE Address->178.238.234.72
List with special IDs not found!

24: ID->2681 Country->PT Address->194.38.140.206
List with special IDs not found!

25: ID->2841 Country->BG Address->193.107.37.135
List with special IDs not found!

26: ID->3021 Country->CA Address->158.69.203.89
List with special IDs not found!

27: ID->3101 Country->US Address->107.191.99.14
List with special IDs not found!

28: ID->3102 Country->US Address->3102.repeater.net
List with special IDs not found!

29: ID->3103 Country->US Address->bm.ham-dmr.org
List with special IDs not found!

30: ID->3108 Country->US Address->64.94.238.196
List with special IDs not found!

31: ID->3109 Country->US Address->63.251.20.56
List with special IDs not found!

32: ID->3119 Country->US Address->74.91.118.233
List with special IDs not found!

33: ID->3129 Country->US Address->hose.repeater.net
List with special IDs not found!

34: ID->3341 Country->MX Address->72.1.241.232
List with special IDs not found!

35: ID->4251 Country->IL Address->109.226.57.187
List with special IDs not found!

36: ID->4501 Country->KR Address->bm.dv.or.kr
List with special IDs not found!

37: ID->5021 Country->MY Address->188.166.231.62
List with special IDs not found!

38: ID->5051 Country->AU Address->125.63.63.112
List with special IDs not found!

39: ID->6551 Country->ZA Address->154.66.196.131
List with special IDs not found!

40: ID->7141 Country->PA Address->45.33.8.38
List with special IDs not found!

41: ID->7241 Country->BR Address->104.41.43.207
List with special IDs not found!

curl -s -f -L --retry 5 --max-time 60 --retry-delay 5 'http://www.dmr-marc.net/cgi-bin/trbo-database/datadump.cgi?table=users&format=csv&header=0' | perl -pe 's,<br/>,,' >dmrmarc.tmp
awk -F, -f insert_nick.awk <dmrmarc.tmp >dmrmarc2.tmp
curl -s -f -L --retry 5 --max-time 60 --retry-delay 5 'http://registry.dstar.su/reflector.db' | perl -pe '$_ = "" if ( $. == 1 ); s#@#,#; s#@.*#,,,,,,#' >reflector.tmp
awk -F, -f combine.awk custom.csv special.tmp fixed.csv dmrmarc2.tmp reflector.tmp | iconv -c -f UTF-8 -t ascii//TRANSLIT  | sort -t ',' -k1g,1 | awk -F, -f strip.awk >stripped.csv	
make[2]: Leaving directory '/home/pi/md380tools/db'
make[1]: Leaving directory '/home/pi/md380tools/db'
"make" -C db stripped.csv
make[1]: Entering directory '/home/pi/md380tools/db'
make[1]: 'stripped.csv' is up to date.
make[1]: Leaving directory '/home/pi/md380tools/db'
wc -c < db/stripped.csv > user.bin
cat db/stripped.csv >> user.bin
./md380-tool spiflashwrite user.bin 0x100000
'Device' object has no attribute 'manufacturer'
Makefile:91: recipe for target 'flashdb' failed
make: *** [flashdb] Error 1
pi@raspberrypi3:~/md380tools $

travisgoodspeed, 2017-06-03:
Closing as a duplicate of #774.

closed 2017-06-03 by travisgoodspeed

jonnycastaway, 2017-06-19:
Hello,
why closing this issue?
I have the exact same problem:

'Device' object has no attribute 'manufacturer'

Under Linux it's all fine but under Windows this problem comes.

jonnycastaway, 2017-06-19:
Ok, Problem solved!
The old pyUSB Version that is referenced in the README.md was the Problem.
Installed new Version and Voila!

roelandjansen, 2017-06-19:
maybe you could send the information/change or create a  PR to fix the docs?