travisgoodspeed/md380tools #777: USB protocol for database?

farnouche, 2017-06-05:
hello everybody,

who can say me where find the communication protocol between TYT and computer through USB for the user database implantation?

I would try to make a software on Windows OS :-)

thank you

travisgoodspeed, 2017-06-05:
Tytera's CPS works on Windows by re-using the STMicro Cube SDK's example DFU code, patched to add their own commands.  They also re-use STMicro's driver, USB VID, and USB PID.

You'll find the device side of the source code as `./applet/src/usb.c`.  From a higher level perspective, we are hooking the USB Device Firmware Update protocol to add our own commands.  Be sure to query the Manufacturer String first, as that's what installs our USB stack hooks on the device side.

I'll draft up some proper documentation of our protocol in the Wiki and close this issue when it's ready.

KG5RKI, 2018-02-09:
Link to my library I made from reversing the stock drivers and writing simple wrappers.  It can be used in place of libusb on windows.  

https://github.com/KG5RKI/dfulib