travisgoodspeed/md380tools #778: Squelch

jasons1150, 2017-06-08:
I am having an issue on analog to me it sounds like the squelch is too tight even on normal. The LED lights up then it's receiving a signal and depending on how I wave the radio around or hit the monitor button it will allow the audio through. Pl is right

MeestorX, 2017-06-22:
Could this be the cause of the scanning bug?
https://www.youtube.com/watch?v=DtE1NyeDpsg

jasons1150, 2017-06-22:
No I don't scan with it

MeestorX, 2017-06-22:
> No I don't scan with it
So... not related, you think?

jasons1150, 2017-06-22:
Nope to me it sound like the squelch is too tight. I cant even turn the squelch off and have it by ctcss

jasons1150, 2017-06-22:
Or maybe I can I'll have to try that

travisgoodspeed, 2017-06-22:
Can anyone reproduce Jason's problem on their own radio?

MeestorX, 2017-06-22:
I can't reproduce that problem, I'll start a new thread for my particular bug.

jasons1150, 2017-06-22:
My Kenwood tk8180 and uv5r allow the squelch to be set to a number

jasons1150, 2017-06-22:
The signal from the repeater maybe a little on the weak side but when I hold the monitor button open it actually sounds good

travisgoodspeed, 2017-06-22:
Regardless, this is a feature request and not a bug in our firmware patches.  Closing issue.

(Sorry neighbor; we can't fix all of Tytera's imperfections.)

closed 2017-06-22 by travisgoodspeed

jasons1150, 2017-06-22:
Ok thanks

jasons1150, 2017-06-22:
Where can I get you firmware I don't think I'm using your