travisgoodspeed/md380tools #786: Errno 110: Operation timed out

jamccormick, 2017-06-17:
I keep having this problem while trying to run `make flash` with an MD-380 and the no-GPS firmware.  It won't communicate, and my radio is now bricked--it does not boot, and it does not have firmware.  

I've attached a dump of the program output when I run `make flash`.  Any advice would be appreciated.

Additionally, I get the same timed out error when trying to run md380-tool dmesg.  I can't get information in or out of the radio.

##Support materials

* The output of `make flash`: [output.txt](https://github.com/travisgoodspeed/md380tools/files/1082771/output.txt)
* The output of `md380-tool calibration: [calibration.txt](https://github.com/travisgoodspeed/md380tools/files/1082773/calibration.txt)
* SPI dump: `Dumping SPI Flash to 'spidump.bin'.  The device has no langid`

I can't get any other information--it isn't there.  I'll get pipe errors, which seems to indicate that the file it's looking for isn't there.

roelandjansen, 2017-06-18:
Reflashing with windows one time does not work?

To me it sounds like your build env is b0rked.

DL2MF, 2017-06-18:
Check your programming cable at the USB-connector side, looks like broken.

DL2MF, 2017-06-20:
@jamccormick @travisgoodspeed This is no sw issue, flashing MD-380 works fine, like just confirmed with a unit. May be closed.

jamccormick, 2017-06-20:
Confirming: not an issue with software. Close it. 

Replacing the cable fixed the problem.

closed 2017-06-20 by travisgoodspeed