travisgoodspeed/md380tools #787: Audio Mute Issue when Scanning enabled for Analog Channels

MeestorX, 2017-06-22:
Set 2 radios are set to the same analog channel so one can transmit to the other.
With Scan OFF on the transmitting radio, when you press PTT, the receiving radio un-squelches immediately and the audio is heard immediately on the receiving radio.
With Scan ON on the transmitting radio, when PTT is pressed, the receiving radio un-squelches immediately as before, but the audio on the receiving radio is muted for a second or two.
If you then re-key the PTT quickly enough, the audio is not muted on the receiving radio and is heard immediately.
It does not matter if the receiving radio is in Scan mode or not.
It does not matter if the scan list has any entries or not.

These videos illustrate the problem.
https://www.youtube.com/watch?v=APJJd47zEWI&list=PLRN1njFTTeD-RqyKqwyXnh9jYpNwaKoWF

In the 1st video, I was keeping the radios near each other so that the you can see the screens and I was using the audio feedback (squealing) to show the audio muting action, instead of talking into one radio remotely. The audio feedback is a good indication of how long the audio is muted when the transmitting radio is in scan mode.
The 2nd video demonstrates the problem using my voice instead of audio feedback. You will hear when I enable Scan, the first few numbers are cut off. In this video the radios are further apart and not overloading the receiver, but you can't see both radios to see their screens.

MeestorX, 2017-06-22:
In both.

travisgoodspeed, 2017-06-22:
I'm sorry, neighbor, but your best bet is probably to wait for the official firmware to fix this.  We haven't reverse engineered much of the squelch or scan code, and it's unlikely that we'll be patching that portion in the near future.

Closing issue.  Devs, feel free to reopen and self assign if you'd like to adopt this.

closed 2017-06-22 by travisgoodspeed