travisgoodspeed/md380tools #788: MD380 with Neew Vocoder - flipped screen after update to 13.20

eebydesign, 2017-06-27:
I updated my new out of the box MD380.  On arrival the display orientation was correct, after update to MD380-D13.20 the display is upside down.  Since that update I've tried several other versions of FW only to get the device into a boot-loop or frozen white screen.  13.20 appears to be the only version that allows the HT to function but the display is upside down.  About ready to box it up and send it back.  Anyone have a patch for this issue.  I've seen a patch on Github for the 390 with a similar issue.

Wolf-DL4YHF, 2017-06-27:
So you get a correct image when rotating the radio by 180° (so antenna 
points down), or is the image mirrored then ?
The LCD controller itself supports any combination of rotation and 
vertical/horizontal mirroring.
Maybe the modified the location (in SPI Flash) where the type of the 
display is stored at the factory.
A relatively simple workaround would be holding a certain key pressed at 
power-on to let the experimental firmware know which kind of display is 
in use, and store that information in non-volatile memory.
Most of the required "ingredients" already exist (for example, polling 
the the key at power-on is already used to enable Morse output) ....

73,
   Wolf DL4YHF .

montyshu, 2017-09-02:
In working another issue I loaded 13.20 and had the same result.  There is a newer version, 13.34, that works properly.

closed 2018-05-30 by mach327