travisgoodspeed/md380tools #790: Merge KG5RKI work into mainline

mach327, 2017-07-06:
@travisgoodspeed Ty (@KG5RKI) has some solid work into his fork of md380tools, for both the MD380 and MD2017, and has expressed an interest in merging back. I can test and merge some of the higher level stuff, but the firmware hackery is still out of my depth. 
See also: #789

travisgoodspeed, 2017-07-06:
Excellent!  I've ordered a radio that ought to arrive next week.

DL2MF, 2017-07-08:
Ty? Isn't that the guy, offering the md380tools fw and Travis' android app labeled with his callsign in the version screen and removing his repo from github ...?

mach327, 2017-07-08:
Probably.

This issue is about merging Ty's code back into the travisgoodspeed/md380tools master branch, which, as far as I can tell, would seem to erase any complaints regarding etiquette or licensing. For the record, I haven't met the license terms yet, myself :)

If you have a grief with Ty regarding etiquette, licensing, or in general, I suggest taking it up with him directly rather than here in the issue tracker.

mach327, 2017-07-20:
https://github.com/KG5RKI/TyMD380Tools

travisgoodspeed, 2018-02-06:
Closing this issue as stale.  Please open a github pull request for merges, which will require forking this repository and then applying the minimal necessary changes to add support.

closed 2018-02-06 by travisgoodspeed