travisgoodspeed/md380tools #794: awk overflow now that dmr-marc line count over 65535 lines

KD4Z, 2017-07-15:
It appears that awk has a line count limit.   I can't find hard evidence fingering awk yet, but it seems to be a 16-bit overflow.  Getting an invalid pointer overflow when creating stripped.csv, now that the dmr-marc download is over 64K   The step that combines and de-dups the data is now bonking.

stripped.csv: custom.csv special.tmp fixed.csv dmrmarc2.tmp reflector.tmp
	awk -F, -f combine.awk $^ | $(ICONV)  | sort -t ',' -k1g,1 | awk -F, -f strip.awk >stripped.csv	

combine.awk is a duplicate row reducer, strip.awk seems to just manipulate in an extra field for country.  Some refactoring around the awk limitation is needed.   I'm poking around for a substution now, but figured I'd open this to bring this up to the Team to help knock this out fast.

KD4Z, 2017-07-15:
It seems to be just the combine.awk step that is overflowing.  Since it looks like it's just a file concat step (not a de-dup pass as I first thought) , I changed the target to this less elegant solution and it stopped overflowing.  Odd that the last pass through awk with strip.awk doesn't overflow.

Anyone have any issue with this brute force workaround before I make a PR?

original:
awk -F, -f combine.awk $^ | $(ICONV)  | sort -t ',' -k1g,1 | awk -F, -f strip.awk >stripped.csv

to:
cat custom.csv special.tmp fixed.csv dmrmarc2.tmp reflector.tmp | $(ICONV) | sort -t ',' -k1g,1 | awk -F, -f strip.awk >stripped.csv

travisgoodspeed, 2017-07-15:
We really ought to ditch the awk scripts for something sturdier, but in the meantime, please do send along a pull request for the quick fix.

KD4Z, 2017-07-15:
ok, it's on the way.

sijskes, 2017-07-15:
Proposed solution does not filter duplicates in order.

The original solution is meant to allow local overrides in custom.csv.

Yours provides random selection of either a override or one from the 
remote db.

travisgoodspeed, 2017-09-15:
@sijskes Is this issue good to close, or do bugs remain?

KD4Z, 2017-09-16:
@travisgoodspeed @sijskes   The original author's awk script was overflowing on a Cast of the placeholder in the de-dup array.  This placeholder contained no real value other allowing the element to be created in the array-- it's existence in the array being testable detection of further dups.    I removed the Cast by simply assigning a numeric instead of "x"      I believe it has been resolved for now and should be closed.

sijskes, 2017-09-16:
the patch does remove the override function, where you can override a record manually. 
so i believe it is not a good fix.

sijskes, 2017-09-16:
but the problem has been solved elsewhere, so i close this.

closed 2017-09-16 by sijskes

KD4Z, 2017-09-16:
>the patch does remove the override function, where you can override a record manually.
>so i believe it is not a good fix.
@sijskes @travisgoodspeed    Just how do you come to that conclusion?   The final PR does not remove that intended de-dup functionality at all.   It was solved here, not elsewhere.