travisgoodspeed/md380tools #795: New MD380 and MD390 without GPS firmware available D13.34

jcmurray, 2017-07-15:
I've just received a new MD380 and it came with factory installed firmware version: D013.034. I noticed that "firmware_files.txt" contains only original firmware versions up to and including D013.020.

Would you be able to update md380tools to include support for reverting to this additional new  firmware version? I don't want to flash md3800tools without a backup path to revert the change to the factory supplied version.

Thanks in advance!

mach327, 2017-07-16:
You can always download the firmware you'd like from Tytera and flash it yourself. And if you can find the firmware binary, update here and I'll add it to the list of original firmware files and add it to the makefile(s) so it's easy to flash. I took a glance but didn't see it explicitly called out.

jcmurray, 2017-07-16:
Thanks for the response. There seems to be copy of [TYT-new Vocoder-MD380 and MD390 without GPS-D13.34.bin](https://open-dmr.fr/telecharger/firmware-tyt-md380-md390-sans-gps-nouveau-vocoder-d13-34/) on this link [https://open-dmr.fr/telecharger/firmware-tyt-md380-md390-sans-gps-nouveau-vocoder-d13-34/](https://open-dmr.fr/telecharger/firmware-tyt-md380-md390-sans-gps-nouveau-vocoder-d13-34/). This seems to be a French OpenDMR site. I can't find a copy anywhere else even on the TYT site. I did a test unwrap as a quick verification exercise:
```
john@violet:~/testing$ ~/stuff/radio-stuff/md380tools/md380-fw --unwrap TYT-new\ Vocoder-MD380\ and\ MD390\ without\ GPS-D13.34.bin foo.img
DEBUG: reading "TYT-new Vocoder-MD380 and MD390 without GPS-D13.34.bin"
INFO: base address 0x800c000
INFO: length 0xf3400
DEBUG: writing "foo.img"
```
... which seemed encouraging but wasn't able to verify that it was a true **img** file for D13.34. I'm not familiar enough with the format to distinguish a valid one from a potentially bogus one and I'm by nature suspicious.

askoeho, 2017-07-16:
The new firmware can be found from http://tyt888.com/?mod=download as "MD-380 SOFTWARE 0321"

It would be nice that md380tools would update to this firmware since it should fix TX Admit Criteria (https://github.com/travisgoodspeed/md380tools/issues/120)

martinhpedersen, 2017-07-16:
FYI, my MD390 with GPS came with S013.034.

I've been in contact with TYT trying to obtain a copy of the firmware. They sent me a file, but unfortunately it was the wrong version S013.032. I sent a follow-up requesting the correct version, but they never responded.

jcmurray, 2017-07-16:
Thank you [@askoeho](https://github.com/askoeho), you are absolutely correct, the D013.034 firmware is indeed in the ZIP here [http://tyt888.com/uploadfile/upfiles/20170321134006.zip](http://tyt888.com/uploadfile/upfiles/20170321134006.zip). No idea how anyone was supposed to figure that out from the link title.

mach327, 2017-07-17:
It doesn't appear to me that the Makefiles exist to make flashing back to stock particularly easy except for debugging purposes - i.e. this usage is more than adequately solved with downloading the desired firmware image, and ` ./md380-dfu upgrade path/to/D013.034.bin`. That's actually exactly what the makefile will do when you run `make flash_original_D13.34`, just with more opinions of where the file will be on your hard drive.

I've added it anyway in 
https://github.com/travisgoodspeed/md380tools/commit/af7b151d476b5ae3639fc38f6198326017962cff, since maybe someday it'll be supported, and no one's screamed "no don't do that!" yet. :)

closed 2017-07-17 by mach327

martinhpedersen, 2017-07-18:
TYT sent me the S013.034 firmware today. If anyone is interested in hosting it, I can forward the email so we can add that one as well.

I have not confirmed the integrity of the file, but it is a zip archive containing `TYT-new Vocoder-MD390 and MD-390 with GPS-S13.34.bin`.

N1GHI, 2017-09-11:
Hey martin any possible way you could email me the .034 firmware n1ghi@yahoo.com 
Thanks buddy