travisgoodspeed/md380tools #799: Archive.org firmware files missing?

garydion, 2017-07-23:
Perhaps I'm the first to observe this... or I got lost somewhere along the way.... but the firmware files needed by the script look to be missing from archive.org.

Archive.org states 

> The item is not available due to issues with the item's content. 

When the Make script gets to this line:
`
https://archive.org/download/TYTMD380FW2/TYT-Vocoder-MD380-D13.20.bin => dl/D013.020.bin Makefile_orig:60: recipe for target 'dl/D013.020.bin' failed 
make[5]: *** [dl/D013.020.bin] Error 22
`

...it all falls apart and of course no .bin files are created.

Did the files on archive.org get removed?

Thanks! 73 -Gary, N4TXI

nmoon, 2017-07-23:
I am having the same problem, all builds fail because of this.

travisgoodspeed, 2017-07-23:
Self assigning.  I'll push a fix in a jiffy.

d235j, 2017-07-23:
Looks like archive.org messed something up. Looking into it.

d235j, 2017-07-23:
@travisgoodspeed if you think it wouldn't be too risky to e.g. store them in a github repo or on github pages, we can do that.

travisgoodspeed, 2017-08-31:
Fixed by @d235j.  Closing issue.

closed 2017-08-31 by travisgoodspeed