travisgoodspeed/md380tools #810: Hash button not anymore possible to doubleclick for talkgroup quick entry s13.020 30.08.17

mike6000, 2017-08-31:
Hash button (#) not anymore possible to doubleclick for talkgroup quick entry 
Md380Tools KK4VCZ s13.020 30.08.17

travisgoodspeed, 2017-08-31:
I'm sorry @mike6000, but this report is a bit vague.  Can you please explain in more detail what the problem is, and what you expect the behavior to be?  On what versions was it possible?

mike6000, 2017-08-31:
Hi Travis. 

TyteraFlashtool was used to download and flash the latest fw approx two weeks ago (around 15 Aug 17). Unfortunately I didnt note down the version and I have owererwritten that fw now. However it was quite current but not the latest one (two weeks earlier than todays date. Todays date is 31 Aug 17).

"##" key feature:
That fw had a nice feature. When you pressed the "#" (hash) key the new menu was shown. 
When pressing "##" the hash key two times quickly the user was taken directly to the entering of talk group. This makes it much easier to change talk group. 

Now you have to press the "red hangup key" to enter the menu. However no double press of the red hangup key takes you to the change talk group entering menu item.

Pls add back this functionality. I know many ppl started to use that.

travisgoodspeed, 2017-08-31:
That feature is unique to the KG5RKI fork of md380tools, and it does not exist in the main branch.  Please contact him for this problem.

Closing issue.

closed 2017-08-31 by travisgoodspeed