travisgoodspeed/md380tools #819: Timeslot issue

KE7EVP, 2017-09-23:
Correction:
When someone comes up on a ts1 talkgroup they come up on all ts1 talkgroups. also
when someone comes up on a ts2 talkgroup they come up on all ts2 talk groups.

travisgoodspeed, 2017-09-23:
I really don't understand this bug report.  What is going on?

danny1990, 2017-09-24:
It's nothing to do with timeslots it's just how DMR works.

The only way to do this is to remove all talkgroups from your radio, disable promiscuous mode and change the talkgroup manually each time via the red button menu -> Tk Grp. If in future you want to listen to more than one talkgroup rather than adding the talkgroups back you can enable promiscuous mode from the menu "Utilities -> MD380Tools -> DMR -> Promiscuous".

The only problem is that you will loose the feature of talkgroup names like "World Wide Eng" etc it will just show the talkgroup number instead.

closed 2017-09-24 by travisgoodspeed