travisgoodspeed/md380tools #820: Radio reboots on earing some TG / Stations

pjao, 2017-10-01:
Don't know if someone haver experience something like this. 

This occurred over repeater and hotspot. Md380 just enters in reboot since is an active transmission on tg7229. The only way that radio don't start rebooting is enter in monitor 3.

With other monitors and screens, it enters in reboot with active transmission.

Any ideas?

travisgoodspeed, 2017-10-01:
Exactly which firmware revision are you running?  It should show you on the startup screen.

pjao, 2017-10-01:
Sorry, last available in here, just compiled.

travisgoodspeed, 2017-10-01:
Are you using scanning mode?  Does the behavior go away if you turn off the Call ID feature?

pjao, 2017-10-01:
Not using scanning mode. 

"Call ID feature", do you mean Show Calls? Try all the options, even "CPS only".

travisgoodspeed, 2017-10-01:
Does the stock firmware have the same problem?  Can you find any setting that prevents it from crashing, to help narrow down where the bug might be?

pjao, 2017-10-01:
Don't have tested with stock firmware yet. The only way that i've discover that it doesn't crash is with monitor 3 open.

pjao, 2017-10-05:
Sorry about the delay, just able to test today.

Tested with the base firmware D013.020 and D13.33 with no problem.

Also have tested with an old tools version from about 2017-05-20 and that as already the problem.

The only option that it doesn't crash is using netmon3. Tested with default options, and several other setting and always the same thing...crash...crash...crash.

Another information is that LastHeard (netmon4), LH channel (netmon5) and  Rx Channel (netmon6) are clean, no data/information on it after RX end and existing netmon3.

What does netmon3 do that blocks the crash? Something like not writing current call information that goes filling also the lastheard? That is the only info common to main screen and other monitors.

travisgoodspeed, 2017-10-05:
If it varies by the netmon page, it's probably a `strcpy()`  bug, format string, or something similarly embarrassing.  I'll take a look soon.

pjao, 2017-10-05:
Another information is that in the same TG, only IDs starting with 7 makes the radio crash. 

There where station from USA started with 3 with no problem. You can look at history of TG 7227 on Brandmeister.

Thanks Travis!

travisgoodspeed, 2017-10-05:
Thanks for the 7 detail.  That will help me reproduce the bug locally on a simplex frequency.

Could you try it on a radio with no Call ID database?  (Not just disabled, but no database at all.)

pjao, 2017-10-05:
How to clean the database? Writing an empty file doesn't do it...

travisgoodspeed, 2017-10-05:
Maybe a file of `FF`?  It's been a long time and I don't have the code handy.

pjao, 2017-10-05:
I think i've found it! 

Someone else problem... Calculating the file size... First line as wrong information. I've fixed it and after that all looks to be fine.

I'm using stripped.csv from Wilfred Pontfoort and uploading it with flash tool from KG5RKI [image]

But the size is not correct calculated and make this error.

Sorry about all this trouble, hope this can help someone else.

travisgoodspeed, 2017-10-05:
Not my circus, not my monkeys.

closed 2017-10-05 by travisgoodspeed