travisgoodspeed/md380tools #822: experimental/promiscuous 4eba047

KE7EVP, 2017-10-15:
Experimental can't be disabled, promiscuous can't be enabled in new build.

travisgoodspeed, 2017-10-15:
Please include more detail.  I don't understand what you are reporting.

KE7EVP, 2017-10-15:
It's an md380In the menu experimental mode can't be disabled and promiscuous mode can't be enabled

ztechire, 2017-10-15:
Hi, I would have appear to have a simila problem, Eithe promicuous is not working or squelch is not opening. I am not receiving anything.

travisgoodspeed, 2017-10-15:
Confirmed the issue.

ztechire, 2017-10-15:
Great.

ztechire, 2017-10-19:
Update, I have now got scanning and promiscuous mode working but only if the RX & TX freqs are not the same. Is this the way it is meant to work ?

DL2MF, 2017-10-20:
I'll have a look on this soon while preparing the next build.

PR #826 (2017-11-07): several menu state fixes
this also fixes #822
commit 9809c600 (2017-11-07): Merge pull request #826 from stefansaraev/menu-fixes

several menu state fixes. Close #822.
closed 2017-11-07 by travisgoodspeed