travisgoodspeed/md380tools #831: Option Mic gain does not work properly.

PD1LOI, 2017-11-25:
There is a problem with Mic gain, If Mic gain has never been activated then nothing is wrong. But as soon as you activate Mic gain 3db or 6db, the sound becomes duller. If you turn the Mic gain off then the Microphone will not work anymore, then you will hear nothing after 1 second. This is an external microphone.

DL2MF, 2017-12-09:
May be you'll be so kind to let us know which radio and vocoder version you are using?

PD1LOI, 2017-12-11:
Several users have experienced this in both the MD380 as the MD390 New
vocode.

travisgoodspeed, 2017-12-11:
Is this unique to the patched firmware, or does the original firmware have the same problem?

PD0DIB, 2017-12-11:
Mic gain is only in custom firmware, not in the stock version. Don't know
who did the pull request, sits there for a while.

====================
met vriendelijke groet | Kind Regards,
Rob van Rheenen

DL2MF, 2017-12-29:
I'm using MD390/G with new vocoder, didn't notice or could reproduce this issue so far. 
Can you please check on netmon3 screen for values of:
FM_DEV load=28
FM_REG load=E0
what values these registers have before/after the issue has been reproduced and repowering the radio.