travisgoodspeed/md380tools #832: UserDB over 5 MB

PD1LOI, 2017-11-30:
Mike any time one now the userdb. bin load goes in you firmware then it goes wrong. Apparently the firmware yet accommodate more than 5 MB data for the userdb.

PD1LOI, 2017-11-30:
Wrong person.

PD1LOI, 2017-11-30:
Mistake of mine.

DL2MF, 2018-01-01:
Has been already increased up to 7MB by #827, maybe closed.

travisgoodspeed, 2018-01-01:
Closing.  Before it gets much larger, we probably ought to look at compressing the database.

closed 2018-01-01 by travisgoodspeed

KD4Z, 2018-01-01:
Working on that now.   I have a simple abbreviation of the top 10 occuring country names with re-expansion in display.c that has achieved an instant 11% reduction.  This is live in my current Toolkit VM based builds.   However, working with another dev on a brute force compression of similar phrases, that has achieved a user.bin that is 58% of original size, but with a long compression time.  So, it's still a work is in progress.

travisgoodspeed, 2018-01-01:
Cool.  If the compression time needs to be long, we can mirror a precompressed database.