travisgoodspeed/md380tools #833: continuous transmission when password is enabled and button is pressed

nf112-de, 2017-12-11:
Hello,

first of all thanks for the work on improving the md380tools!

Issue:
When power on password is enabled and device is locked (after power on),
it is possible that device switches to continuous transmission when a TX button
is pressed before the device is unlocked.
After unlocking the device, it switches to continuous transmission that can only be stopped by immediate power down.

Steps to reproduce:
- enable power on password in CPS
- program radio
- power on radio (password screen shows up)
- press a TX button (for example call button (1750 tone))
- (nothing happens, no transmission)
- enter password and unlock device
- device switches to continuous transmission (can only be stopped by power down)

Tested on MD390 with GPS

Regards,
Niklas DO5NF

travisgoodspeed, 2017-12-12:
Is this issue unique to md380tools, or does it also happen with the official firmware from Tytera?

nf112-de, 2017-12-14:
Hi,

can't verify that at the moment, I'm short on time.

Regards

closed 2018-02-06 by travisgoodspeed