travisgoodspeed/md380tools #837: license clarification

osmoc, 2017-12-29:
The humorous description of license in readme is nice but it would be much better tofacilitate code reuse by explicitly setting the license via usual "copying" file or using https://spdx.org/ tags. It's also dubious that GPL code can be redistributed with additional "2 beers" requirement so it would be better to explicitly clarify which files are under GPL and which are not (if any).

travisgoodspeed, 2017-12-29:
Because we statically link against Tytera's proprietary radio firmware, GPL is not an option for the firmware.  As `README.md` states, only the host-side Python tools such as `md380-dfu` and `stm32-dfu` are available under the GPL.

A far bigger problem than static linking is that while I've received a beer or two for this project, I've never received the two liters required by the license.  Not one of the thousands of users is in compliance!

To simplify the licensing, I'll write to Great Scott Gadgets and ask for their blessing to move the Python tools over to the beer license.  They've been pretty cool about not suing me in the past.

73 from Reno, Nevada,
--Travis KK4VCZ

travisgoodspeed, 2018-01-21:
I purchased the copyright to the old DFU code from the Ubertooth project this weekend, and the python scripts are now unified under the beerware license.  Closing issue.

73 from Baltimore, Maryland,
--Travis

closed 2018-01-21 by travisgoodspeed