travisgoodspeed/md380tools #839: No audio level on the screen when you turn the volume knob.

PD1WP, 2017-12-31:
When I turn the volume knob, the volume works but does not show anything on the display. Run with the latest firmware version of Travisgoodspeed that I downloaded from the PD0ZRY website.
I hope your team can solve this. Thank you for your work and wish you and your team a fantastic 2018. 
https://wappiesoft@wappiesoft.stackstorage.com/s/jRvQKWHCQvFXoRK5
73.

PD1WP, 2017-12-31:
The link is a link to a video in which I show the problem. Volume works fine, but does not show anything on the display.

stefansaraev, 2017-12-31:
volume screen was removed in 5cc2f998 @DL2MF perhaps it got in by accident or was it intentional?

PS: I am happy with the change :)

DL2MF, 2018-01-01:
The nasty screen volume bar has been removed like documented in the sources. Many users noticed the bar without turning the volume knob. It's also absolutely useless, cause the level often toggles/showing unreproducable values/settings.

HNY @all #md380tools friends!

travisgoodspeed, 2018-01-01:
Closing this issue.  Good riddance to the volume bar.

closed 2018-01-01 by travisgoodspeed

stefansaraev, 2018-01-01:
fyi this may have an unwanted side effect to backlight timer: after using the volume knob, screen does not always go dark after specified timeout

happy new year to everyone!