travisgoodspeed/md380tools #841: MD390 - Screen reversed and upside down

vk2crt, 2018-01-04:
Hello there all...

Please forgive me if this is not the right place to post this issue.

I have a MD390 running MD380tools Ver. : 2018-01-04, CP Ver. : V01.37.

The screen is reversed and upside down. See picture attached.
[image]

I also have a MD380 running MD380tools Ver. : 2018-01-04, CP Ver. : V01.27. And the screen on the MD380 is correct.

Both have been updated via my linux box from the GitHub directly.

Is there a change I need to make to get the MD390 screen to display correctly?

The difference between the two radios, as far as I can see, is that the MD380 has CP Ver 01.27 and the MD390 has V01.37

I am happy for advise and suggestions as needed.

Col Thompson
VK2CRT
Sydney Australia

DL2MF, 2018-01-04:
Did you already try this fix, described a few lines below your issue report?
https://github.com/travisgoodspeed/md380tools/issues/788

vk2crt, 2018-01-04:
Hello and thanks for your reply.

Sorry but no the D13.34 firmware does not fix the inverted and back to front screen. I copied the TYT-new Vocoder-MD380 and MD390 without GPS-D13.34.bin file to the md380tools\firmware\bin\ directory, and renamed it D13.34.bin. I then did a make flash_D13.34.

I also used a third party windows update utility on my windows box with the same result.

So I may be doing something fundamentally wrong. Is there a way to change the CP ver from V01.37 to V1.027? An I looking in the wrong place?

Col
vk2crt
Sydney Australia

DL2MF, 2018-01-05:
The CPS doesn't affect your FW issue. Use the genuine FW for your model and flash with the TYT Update tool, to find the correct FW for your model, then you may try md380tools again based on that build.

Olimatou, 2018-01-11:
HI,

This is set by pressing wrong buttons when turning the radio on. I can demonstrate this with my md398 attached picture below...

Simply turn your radio off then maintaining PTT switch + Side Key 2 pressed, turn your radio on, then it will round robin on screen rotating settings. Repeat the procedure to get your screen on the right setting.

Regards.

[image]

vk2crt, 2018-01-11:
Thanks to all that have replied.

Try as I might, the screen reversed and upside down issue remains. The only firmware that works for me on my MD390, is firmware-2017-05-28-OLD.bin.

As for Olimatous advise. Yours is the ONLY post that says press power and button two that I have seen on the greater internet. I can only assume you are using a specialised firmware variation that is particular to your MD398. It does not work for me.

I will say this though. The md380 tools from the Travis Goodspeed GIT does work on an MD380. It seems, and this is from me, that it's flaky when used on other models / clones of the MD380.

Regards and 73 to all.

Col
(aka VK2CRT)
Sydney Australia

Olimatou, 2018-01-11:
Hi,

I use the original firmware. I didn't received my programming cable yet. I ordered a GD-55+ (plus version) to start my researches on these two different firmwares and to get the md-398 model modded for slot 2 and full tier II support...

You should try combinations of pressed keys when turning your radio on...

Regards.

vk2crt, 2018-01-11:
Thanks for your reply.

It does not work for me and my MD390.

Regards

Col
(aka VK2CRT)
Sydney Australia

closed 2018-01-11 by vk2crt

Olimatou, 2018-01-11:
Well, maybe you should try on original firmware... If this is working, then try to rotate your screen prior flashing modded firmware to see if this affect final display orientation... Won't know if it will work, but this is experiments to try before giving up ^^

rogerclarkmelbourne, 2018-03-27:
Someone sent me a MD380 that also has the screen inverted problem

Checking the firmware version, its showing

D013.014

So its still running the official firmware
I'll try installing this firmware but I thought it would be best to get the screen the right way up first..

BTW. I tried using PTT + SK2 on boot, but it doesn't do anything

Not sure if its related but neither this radio or my MD380 will connect to the official CPS, either before or after installing the new firmware. But DFU works fine on both

ryangd55, 2018-04-23:
i had the same problem i fix it using  using PTT + SK2 on boot, keep it held down till you see the boot screen it wiil take 3 times doing this to rotate you're screen 
[image]
[image]
[image]

DL2MF, 2018-04-23:
This is not GD-55 support group, the thread opener was reporting a MD-380 issue (see pictures of the radio! green key left, 2 arrows, red key right).
MD380tools project is not supporting GD-55 or any of the firmware versions of this radio.