travisgoodspeed/md380tools #844: md380-dfu upgrade broken

ZeroChaos-, 2018-01-18:
From make flash_S13:
./md380-dfu upgrade applet/experiment.bin
argument 2: <type 'exceptions.TypeError'>: wrong type
Makefile:74: recipe for target 'flash_S13' failed
make: *** [flash_S13] Error 1

this was working fine last I tried it, but appears to be broken now....

ZeroChaos-, 2018-01-19:
Note, it seems to work on ubuntu but not on debian stretch, so I'm assuming some version is messed up or some line of code isn't portable across ?

ZeroChaos-, 2018-01-19:
I don't know if the world hates me, but this is an ubuntu xenial chroot:

./md380-dfu upgrade applet/experiment.bin
[Errno 75] Overflow
Makefile:74: recipe for target 'flash_S13' failed

both of these different failures are chroots on my gentoo box.  I wouldn't expect that to matter, but I'm stating it for full disclosure

ZeroChaos-, 2018-01-19:
The error appears to be literally random:
```
root@gato:~/md380tools# ./md380-dfu upgrade applet/experiment.bin
[Errno 5] Input/Output Error
root@gato:~/md380tools# ./md380-dfu upgrade applet/experiment.bin
[Errno None] Other error
root@gato:~/md380tools# ./md380-dfu upgrade applet/experiment.bin
[Errno None] Other error
root@gato:~/md380tools# ./md380-dfu upgrade applet/experiment.bin
[Errno None] Other error
root@gato:~/md380tools# ./md380-dfu upgrade applet/experiment.bin
[Errno 75] Overflow
root@gato:~/md380tools# ./md380-dfu upgrade applet/experiment.bin

root@gato:~/md380tools# 
```
I am pretty confused.  nothing wierd in dmesg and the same cable/radio works fine on a real ubuntu box but not an ubuntu or debian chroot on my gentoo box.

travisgoodspeed, 2018-02-06:
As no one has been able to reproduce this, I'm assuming that something is wrong with the chroot environment.  Please send support requests to the mailing list and not to the issue tracker.

closed 2018-02-06 by travisgoodspeed