travisgoodspeed/md380tools #847: backlight staying on

nivex, 2018-02-02:
Since upgrading to 77865a0, the backlight on my 380 is randomly staying on. 
Menu -> Utilities -> MD380Tools -> Display -> Backlight
Level Low is set to "0 (off)"
Level High was "9 (bright)" but I have also tried it with "5"
Backlight Tmr is "10 sec"

Of possible note, when I go into the Level High menu, none of the radio buttons are set, but the menu bar highlights the current setting.

Sadly, when I say "randomly" I mean I haven't been able to isolate the conditions under which this occurs. I'll set the rig down, glance back some minutes later and notice the display is lit. I stare it down for 10+ sec to make sure it's not a coincidence of an event having recently lit it up. Most times I can tap a button and it will then go to sleep in 10 sec.

stefansaraev, 2018-02-03:
hint https://github.com/travisgoodspeed/md380tools/blame/master/patches/d13.020/patch.py#L45-L48

KD4Z, 2018-02-03:
@nivex   I have a tweak in my VM based build setup that allows unpatching this patch.  You would have to install my VM, enable the tweak in the screenshot, and then build the firmware using the commands in the VM.  The backlight timer is working after reenabling the mic bargraph.  Project his here: https://github.com/KD4Z/md380tools-vm    It's a lot to go through, I know.  You might find other interesting bits come along as well.  
[image]

nivex, 2018-02-03:
@stefansaraev There's a lot going on in that commit. It might be useful to know why the volume bar was patched out. I would not be opposed to that functionality returning, especially if it fixes the backlight bug.

stefansaraev, 2018-02-05:
@nivex well, see #839 it is a feature, not a bug :)

DL2MF, 2018-02-06:
I noticed the sideeffect from #839 but currently I'm very busy with my job. To less time to work on the tools during the next weeks.

nivex, 2018-02-10:
Further observation: When the radio is in this condition, the display is not being updated at all. The RSSI bargraph does not bounce, and when in scan the Mode display does not update.