travisgoodspeed/md380tools #848: MD390GPS Shipped mid Jan 2018 will not run md380tools posted on many sites.

Lakeperson, 2018-02-03:
First, I am not a programmer.  I have to rely on other sites that compile and post their files to download, but none seem to work on my or my friends MD390 with GPS that were both shipped in January 2018 about 10 days apart.  We both have independently tried with no success to find a version that works correctly and have had to revert back to the TYT supplied SO13.032 CP Version VO136 to use our new radios.  Some versions have missing pixels and the keys are locked, others reboot on any receive, one cycles a reboot continuously.  Any help would be appreciated.

f1rmb, 2018-02-03:
Hi,

   I have experienced the same problems (reboot on receive and random reboots) with a new 390GPS, after two weeks of usage.
Even reverting to original FW didn't solved the problem. So, I just returned the transceiver, and got a brand new one replacement, no problem since then, maybe they produced a bad batch.

Cheers.
---
Daniel

Lakeperson, 2018-02-03:
PROBLEM RESOLVED when I used the TyTera Flash 1.8 Tool selecting the TYKDToolz  Firmware and then most importantly then running the  USERDB  Download Update.  Apparently there is something going on where they both have to be updated together to make it work.  Everything seems to be working as expected after updating both of them.   Thanks to John K5SJW for this info.

closed 2018-02-03 by travisgoodspeed

PD0DIB, 2018-02-03:
Advice: do not use multiple (parts of) Tools mixed together:
If you use the KD4Z Toolkit, use only that toolkit for download, compile
and flash the firmware AND the userDB.
TIP: use only tools which are sourced in the public domain, preferable
GitHub; so problems can be solved by analyzing the code by specialists.

====================
met vriendelijke groet | Kind Regards,
Rob van Rheenen