travisgoodspeed/md380tools #849: MD380 seems to lock up after boot but USB commands still work

mich181189, 2018-02-05:
I just built and flashed (the prebuilt version on the website seems to have the same problem) the latest version from Git, using:

make flash
(reboot)
make updatedb_eur flashdb

however now the radio seems to boot up normally but doesn't respond to key presses. Turning it off writes "powering down" over the display (without erasing the previous text) and things like dmesg and grabbing screenshots via USB work fine.

dmesg shows:

> Starting main()
> Performing AES startup check.
> aes_loadkey (0x20014a80):  f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed 5e 9c df 14
> aes_cipher(0x20014a60);
> IN  : a6 80 81 7d 2d 00 35 00 11 51 36 31 32 38 38 36
> OUT : 2f e4 82 a2 f6 53 3a 56 ff 8c 32 0e 00 a2 07 b6
> Startup AES check failed.  Attempting to forge the results.
> *0x2001d39b = 0x00
> can't find dnld_tohook_addr
> Size is known to be 01000000

mich181189, 2018-02-05:
it gets weirder:

travisgoodspeed, 2018-02-06:
Could you please tell me the hardware model number of your radio?

The screen is probably rotated in stock D013.020 because of that, and you can rotate the screen back to the normal orientation once you get md380tools working or by flashing the most recent Tytera firmware.  I don't know what would cause the buttons to fail, and that might be a manufacturing defect.  Do the buttons work in Tytera's latest firmware?

mich181189, 2018-02-06:
Tyt’s latest firmware is the right way up and some buttons work (such as locking the keys) but the menu button doesn’t.

md380tools used to work fine but the same version I’ve been using since last May now locks up. 

I figure it’s either some data corruption or reflashing it with the newest version stressed it in some way and broke it.

mich181189, 2018-02-06:
To be clear, I’m now fairly sure this isn’t a bug as such - it just seemed that way because it was triggered by an update

travisgoodspeed, 2018-02-06:
You might try wiping the codeplug to a known-good image, in case that was corrupted.  There's all sorts of janky parsing from the official firmware that might crash with a weird codeplug.

Closing issue.

closed 2018-02-06 by travisgoodspeed