travisgoodspeed/md380tools #851: MD380 tools for new firmware: TYT-new Vocoder-MD390 and MD-390 with GPS-S13.34.bin

palkoc, 2018-02-06:
Hi all,
is there any plan to patch the newest MD390+GPS firmware: TYT-new Vocoder-MD390 and MD-390 with GPS-S13.34.bin ?

Thanks.

Pavol

travisgoodspeed, 2018-02-06:
Feature requests need to go to the mailing list, and not to the issue tracker.  This is explained in the issue template, which you were politely asked to read before submitting your issue.

(To answer anyways: Our codebase can currently automate the porting of function symbols, but there is still significant labor required to port all of the instruction patches.  We will probably not support newer versions until there is a real need.)

closed 2018-02-06 by travisgoodspeed

palkoc, 2018-02-06:
thanks Travis for the answer.