travisgoodspeed/md380tools #890: Compatible Hardware

i2bator, 2018-10-03:
Its almost the end of 2018.

Anyone could please clarify these 2 questions:
"What radios are still fully or partially compatible with this firmware?" (please see the list below)
"Are there any compatibility issues with particular sellers online (same model but slightly different hardware or stock firmware, or New Vocoder/ Old Vocoder?

'73

Following is a list of possible DMR models in question:
____________________________
TYT MD-380 UHF
TYT MD-380 VHF
TYT MD-380G VHF+GPS
TYT MD-380G UHF+GPS
____________________________
TYT MD-UV380 (UHF+VHF)
TYT MD-UV380 (UHF+VHF+GPS)
____________________________
TYT MD-390 UHF
TYT MD-390 VHF
TYT MD-390G VHF+GPS
TYT MD-390G UHF+GPS
____________________________
TYT MD-UV390 (UHF+VHF)
TYT MD-UV390 (UHF+VHF+GPS)
___________________________
TYT MD-2017 (UHF+VHF)
TYT MD-2017 (UHF+VHF+GPS)
____________________________
Retevis RT3 UHF
Retevis RT3 VHF
Retevis RT3 VHF+GPS
Retevis RT3 UHF+GPS
___________________________
Retevis RT3S (UHF+VHF)
Retevis RT3S GPS (UHF+VHF+GPS)
___________________________
Radioddity GD-77 (UHF+VHF)

?? any other radios??

travisgoodspeed, 2018-10-03:
Close this issue when the list is verified and at the top of README.md.  We should include a list of incompatible models just to be clear.

Scytie, 2018-11-03:
Please add Retevis RT82 to this list, i think it is exactly the same as TYT MD2017.
https://github.com/KG5RKI/TyMD380Tools/tree/md2017
had work on it. But it seem to be some files missing to compile it.
Here are some Links to compiled binaries for the RT82.
https://github.com/KG5RKI/TyMD380Tools/issues/2

travisgoodspeed, 2018-11-04:
None of the dual-band radios (MD2017, RT3S) are supported by this project, only the mono-band UHF and VHF radios.  TyMD380Tools is a hard fork with no version history, and merging it to the master branch will not be easy.

Scytie, 2018-11-04:
When i get my RT82 and get when KG5RKI's version running. I diff his code with this code here for a pull request. But a this moment I need a file I do not have. It is the Header of the unwrapped firmware. It should be 1.3mb - 900kb = 400kb in size.
(See KG5RKI/TyMD380Tools#2)

Scytie, 2019-02-19:
just a status update:
Ty (KG5RKI) uploaded his RT82 Stuff:
https://github.com/KG5RKI/RT82Toolz
Using python = python2 I am able to build a new firmware.
If I find time I will diff this repo and https://github.com/KG5RKI/RT82Toolz to merge these two.

adron51, 2019-05-29:
> 
> 
> Please add Retevis RT82 to this list, i think it is exactly the same as TYT MD2017.
> https://github.com/KG5RKI/TyMD380Tools/tree/md2017
> had work on it. But it seem to be some files missing to compile it.
> Here are some Links to compiled binaries for the RT82.
> [KG5RKI/TyMD380Tools#2](https://github.com/KG5RKI/TyMD380Tools/issues/2)

add retevis rt80 too please

adron51, 2019-05-29:
i can only see dmr id om my retevis would love to see all info like i do on my md380 thanks

driviera, 2019-06-18:
Would it be worth noting on the supported MD380/MD390 models about the various hardware versions and reported incompatibilities, and how to read the hardware version in CPS?

kquinsland, 2019-06-23:
@driviera yes, please! I'm still quite new to this particular scene, but it's incredibly difficult to find good documentation on:

- what the "difference" between `TYMDToolz` and `BetaMDToolz` and `MD380tools` is. The first two are forks of the former that add new features... but none? of the forks was merged back into the upstream!?  

- how to tell if the device you're holding is compatible with any of the custom firmware. the warning [here](http://www.miklor.com/DMR/DMR-Toolz108.php) just says v14 and higher wont work.  My unit has a FW version of `S018.003` and a CP version of `1.06`.  I am assuming that the "14" is in reference to the radio FW version and that 18 > 14 so i am SOL... but that warning was on the `TyMDToolz` page, so only _that_ fork (which, hasn't?) been merged into upstream is? affected?

- Which version of the vocoder do i have? That's something you figure out by looking at the cp? version?

FWIW, i have a `MD-UV390 (GPS)`

PD0DIB, 2019-06-23:
Travis Goodspped and team is the originator of the so called 'experimental
firmware', aka custom firmware.
MD380Toolkit is a shell and a bundle of utility scripts, using the Travis'
firmware as the 'engine' for this Toolkit.
Both are well supported and each add-on or modification / bug-fix is well
tested by the Travis-team before release.
The ToolZ are an earlier off-spring and isn't following the Travis
guidelines.

driviera, 2019-06-23:
> FWIW, i have a `MD-UV390 (GPS)`

These tools are not for the MD-UV390. They are for the MD-380 and MD-390 (with or without GPS; and parallel single-band radios). Many of the features of the MD380Tools are already integrated into the native MD-UV390 firmware.

jimmycapizzi, 2019-07-05:
The MD-UV380 does not have the same features as this custom firmware, like ad-hoc mode, and a support for this radio would be greatly appreciated by me, and the community.

G4AON, 2019-07-16:
I too would love extra features on the MD-UV380

73 Dave

AppliedEM, 2020-01-21:
Bump. Got the dual band version of the tyt 390G and found out the hard way that the dual band wasn't supported: red/green blinkies and no boot. Support for this radio would be awesome.

rogerclarkmelbourne, 2020-04-23:
Could someone let me know whether the encryption on the UV-380/390 is the same , or possibly similar to the MD-380, or whether it would be possible to retrieve a copy of the bootloader from those radios, like Travis originally did?

It does not matter to me whether the MD-380 tools binaries work on these radios, which they almost certainly won't since the baseband IC is not the same.

But I'd like to be able to encrypt and upload my own firmware.

mach327, 2020-04-23:
@rogerclarkmelbourne  I suspect the UV-380 is basically just a MD-2017 in a -380 case. If that's true, you might try Ty Weaver (KG5RKI)'s  fork of md380tools, branch md2017: https://github.com/KG5RKI/TyMD380Tools/tree/md2017

rogerclarkmelbourne, 2020-04-23:
@mach327 

Thanks.  I'll take a look.

I'm thinking of buying a UV-380 (actually a RT3S which I'm pretty sure is the same radio), but I don't want to throw money away on something which the encryption hasn't been broken.

ae14, 2020-05-19:
Is it possible to read and write code plug .rdt files to and from the UV-380 with md380-dfu? I'm not trying to do firmware updates at this stage, just to program my radio. If not is there any other known way to read and write code plugs to a UV-380 without having to use the TYT CPS software on Windows?

mach327, 2020-05-19:
@ae14 [Mr Farnsworth's excellent Editcp](https://www.farnsworth.org/dale/codeplug/editcp/) will do this. Editcp does a great job as a full TYT CPS replacement/clone.

If you're running Mac or Linux, and have Chrome or Chromium, you might also try [my WebDFU programmer entirely in the browser](https://dmr.tools) and let me know how it works for you. It's, uh... not ready for primetime yet, but it Works On My Machines (tm) and I could use the feedback for when I get back to working on it. 

Mine currently won't read codeplugs off the radio and give you an RDT, it just gives an unwrapped bin, but it writes them just fine.

rogerclarkmelbourne, 2021-09-13:
Further to my other dumb questions....

Has anyone extracted the bootloader from the MD-9600 or the UV-380

Actually, I'm sure I can extract the bootloader from the UV-380, but I'm not sure if the tools can be used to extract it from the 9600 and I don't have access a 9600 to try it

VorosEgyes, 2024-07-06:
> Further to my other dumb questions....
> 
> Has anyone extracted the bootloader from the MD-9600 or the UV-380
> 
> Actually, I'm sure I can extract the bootloader from the UV-380, but I'm not sure if the tools can be used to extract it from the 9600 and I don't have access a 9600 to try it

I think i ruined my tyt uv-390. only leds blinking in red (no green). Firmware uploade seems good, but i have only black screen. It could be a boot load problem. @rogerclarkmelbourne did you managed to extract the boot file?

rogerclarkmelbourne, 2024-07-06:
[md9600_bootloader.zip](https://github.com/user-attachments/files/16117594/md9600_bootloader.zip)

I did this ages ago, and I can't remember which version of MD9600 radio hardware I extracted it from. Probably the V4A aks V5 radio hardware version.

If you install OpenGD77 on your radio and then use the Backup MCU ROM function in the OpenGD77CPS you can backup the entire contents of the MCU ROM and the lower 0xC000 will be the bootloader.

I don't have access to any of my radios at the moment, so I can't back them up for you. Possibly someone else can

VorosEgyes, 2024-07-07:
> [md9600_bootloader.zip](https://github.com/user-attachments/files/16117594/md9600_bootloader.zip)
>
>
>
> I did this ages ago, and I can't remember which version of MD9600 radio hardware I extracted it [...]

Thank you! I will try it.

rogerclarkmelbourne, 2024-07-07:
@VorosEgyes 

I'm not sure what you are trying to do. 

Loading a MD9600 bootloader onto a UV380 would cause problems because the encryption key is different between those 2 radios

VorosEgyes, 2024-07-07:
> @VorosEgyes 
> 
> 
> 
> I'm not sure what you are trying to do. 
> 
> 
> 
> Loading a MD9600 bootloader onto a UV380 would cause problems because the encryption key may be different between those 2 radios

I mean I will try reading, searching furthet to find the problem! I think it will be something really easy problem...

smirrom, 2024-12-08:
@VorosEgyes @rogerclarkmelbourne 
Hello, is there any updates to extract bootloader from TYT MD-UV390/380?

rogerclarkmelbourne, 2024-12-08:
If you install OpenGD77 onto the UV380 or 390, and then use the Backup MCU function of the CPS, it will read the entire MCU ROM including the bootloader.

smirrom, 2024-12-09:
> If you install OpenGD77 onto the UV380 or 390, and then use the Backup MCU function of the CPS, it will read the entire MCU ROM including the bootloader.

Unfortunately I can't install OpenGD77. My radio came to me with a proprietary custom firmware which differs from original TYT firmwares, and flashing official FW gives no result (tried many of them), radio stuck in DFU (green-red lights blinking). I think there is a custom bootloader with different FW crypto keys, or different entry point to firmware code. So, I need original bootloader to flash original firmware, and OpenGD77 after all.

> Do you mean the UV380 or the MD380.

My radio model is MD-UV390 Plus (10W).

rogerclarkmelbourne, 2024-12-09:
[bootloader.zip](https://github.com/user-attachments/files/18058919/bootloader.zip)

Why did you radio come with a nonstandard bootloader?

smirrom, 2024-12-09:
> [bootloader.zip] 

Thanks!

> Why did you radio come with a nonstandard bootloader?

It came to me used, it worked in the closed communication system "Veda", this is firmware with improved protection against key reading and encryption (as the developers claim).

> I presume you plan on erasing the MCU and then reflashing using STLink ??
Yes

rogerclarkmelbourne, 2024-12-09:
OK.

The bootloader is the one I just read from my radio.  I can't remember if its the original one that was in the radio, because sometimes I have to reflash the bootloader if I want to debug via ST-Link, but I don't ever remember  making a modified version of the UV380 firmware.

Anyway, that bootloader works fine and will load the official firmware and OpenGD77 etc