travisgoodspeed/md380tools #893: Intro Screen showing "unknown 0" after upgrading FW

geiraage, 2018-12-09:
I havnt upgraded for a while, so I upgraded today D13.020 150c1de 2018-12-09

The introsceen that show 2 lines in the configuration, is replaced with Line 1 "unknown" and Line 2 "0"

I am using MD-380 (UHF) old vocoder.

Tx

geiraage, 2018-12-09:
I tried to going back to original FW - make flash_original_D03
The result was back to normal. I then tried to reinstall FW - make flash.
The error is still there.

travisgoodspeed, 2018-12-11:
Could you please post your codeplug?

geiraage, 2018-12-11:
I posted it, buy answering your email Travis. I don't know if you got it?

travisgoodspeed, 2018-12-12:
No, I think it got lost as your email went to Github instead of to me.

geiraage, 2018-12-12:
Do you have a email Travis? I will email it to you.

travisgoodspeed, 2018-12-14:
Flashing your codeplug to my radio with editcp 0.11.6, I get "Amundsen" / "LA40" as the startup text.

KD4Z, 2018-12-14:
@geiraage , try setting **CoPl Override** to **Disabled** under the Developer menu.  That is the default and seems to be the best selection.

Loria, 2018-12-20:
Hi, 
I encountered a similar issue with powing on the radio and a display of "0" instead of the char strings set in the codeplug. Uploading S13.033 into the radio (see .below) and char strings will get displayed.

KD4Z 
> @geiraage , try setting **CoPl Override** to **Disabled** under the Developer menu. That is the default and seems to be the best selection.
pointed into the right direction. Thank You
Even though disabling **CoPl Override** did not help me. I checked other options there. I tested the reset of the developer config with the menueentry *Reset** inside the Developer Menue. 
A following power up of the radio showed the char-strings set in the codeplug.
Maybe this helps 'geiraage' as well.

My radio is an MD390G(ps) HW Version, which needs at least S13.033 to have a correct display. S13.020 will render the display upside down. Sadly tytera does not provide the S13.033 on their website; or I did not find the correct Archive with it :-)

Thanks for all the work.

vy 73 de Loria

travisgoodspeed, 2020-01-07:
Closing as stale for more than a year.  Reopen if the bug is still seen.

closed 2020-01-07 by travisgoodspeed

geiraage, 2020-08-18:
I corrected this with Developer menu and config reset.