travisgoodspeed/md380tools #897: Make error Fedora 29

mmilleror, 2019-01-28:
I have a fresh install of fedora 29.  I followed the software dependency install list.
I'm getting the following error when I do make flash.

 ../md380-fw --unwrap bin/D013.020.bin unwrapped/D013.020.img
DEBUG: reading "bin/D013.020.bin"
Guessing MD380 for radio model
INFO: length 0xf3000
DEBUG: writing "unwrapped/D013.020.img"
make[4]: Leaving directory '/home/user/code/md380tools/firmware'
make[3]: Leaving directory '/home/user/code/md380tools/firmware'
cp ../../firmware/unwrapped/D013.020.img unwrapped.img
./patch.py
Creating patches from unwrapped.img.
Patching hword at 08033f48 to 46c0
Patching hword at 08033f4a to 46c0
Patching hword at 08033f5e to 46c0
Patching hword at 08033f60 to 46c0
Patching hword at 08033f74 to 46c0
Patching hword at 08033f76 to 46c0
Patching hword at 08033f8a to 46c0
Patching hword at 08033f8c to 46c0
Patching hword at 08033fa0 to 46c0
Patching hword at 08033fa2 to 46c0
Patching hword at 08033fb6 to 46c0
Patching hword at 08033fb8 to 46c0
Patching hword at 0804916e to 46c0
Patching hword at 08049170 to 46c0
Patching hword at 080499f4 to 46c0
Patching hword at 080499f6 to 46c0
Patching hword at 08049cb8 to 46c0
Patching hword at 08049cba to 46c0
Patching hword at 0804a144 to 46c0
Patching hword at 0804a146 to 46c0
Patching hword at 0804abd0 to 46c0
Patching hword at 0804abd2 to 46c0
Patching hword at 0801fed2 to 46c0
Patching hword at 0801fed4 to 46c0
Patching range from 0809aee8 to 080cf754 to FF.
Patching word at 0800c020 to 080f9245
Patching word at 0800c004 to 0809af01
Patching hword at 0809af00 to 4840
Patching hword at 0809af02 to 2100
Patching hword at 0809af04 to 3901
Patching hword at 0809af06 to 4508
Patching hword at 0809af08 to d100
Patching hword at 0809af0a to 483c
Patching hword at 0809af0c to 4700
Patching word at 0809affc to 080f9245
Patching wide string at 080cfcac to 'MD380Tools Ver.'.
Patching string at 080f86c8 to 'Travis Goodspeed KK4VCZ'.
Patching string at 080cfff4 to 'Patched MD380'.
../../md380-gfx --firmware=patched.img --gfx=replacement-font-big-latin.pbm fontreplace
DEBUG: reading "patched.img"
DEBUG: writing "patched.img"
../../md380-gfx --firmware=patched.img --gfx=replacement-font-small-latin.pbm fontreplace
DEBUG: reading "patched.img"
DEBUG: writing "patched.img"
make[2]: Leaving directory '/home/user/code/md380tools/patches/d13.020'
cp ../patches/d13.020/patched.img ./merged.img
python2 merge_d13.020.py merged.img applet.img 0x0809b000
Merging an applet.
Loading symbols from applet.img.sym
Inserting a stub hook at 0800c72e to 0809cdfd.
Patching SysTick_Handler in VT addr 0x0800c03c,
  old value in vector table = 0x08093f1d,
   expected in vector table = 0x08093f1d,
  new value in vector table = 0x080a6275.
  SysTick_Handler successfully patched.
Merging applet.img into merged.img at 0809b000
Merging applet.img into merged.img at 0809b000
../md380-fw --wrap merged.img wrapped.bin
DEBUG: reading "merged.img"
Guessing MD380 for radio model
INFO: base address 0x800c000
INFO: length 0xf3000
DEBUG: writing "wrapped.bin"
cp wrapped.bin experiment.bin
make[1]: Leaving directory '/home/user/code/md380tools/applet'
./md380-dfu upgrade applet/experiment.bin
Device not found
make: *** [Makefile:65: flash] Error 1
I don't have the radio plugged in. I do have a md380 and is currently stock.

Current firmware ver is D014.004
CP ver is V01.30.

Feel free to close if I did something wrong.

mmilleror, 2019-01-28:
Funny enough, make clean dist runs clean as root and my user.
-rw-r--r--. 1 root root 4367640 Jan 28 00:39 callerid.csv
-rw-r--r--. 1 root root  999936 Jan 28 00:38 firmware-2019-01-28-GPS.bin
-rw-r--r--. 1 root root  995840 Jan 28 00:38 firmware-2019-01-28-NoGPS.bin
-rw-r--r--. 1 root root  994816 Jan 28 00:38 firmware-2019-01-28-OLD.bin
drwxr-xr-x. 2 root root    4096 Jan 28 00:38 python
drwxrwx---. 2 root root    4096 Jan 28 00:38 windows

travisgoodspeed, 2019-01-28:
`make flash` tries to flash the radio, and for this the radio needs to be plugged in.

Closing issue.

closed 2019-01-28 by travisgoodspeed