travisgoodspeed/md380tools #903: Extract AMBE from firmware as standalone library

smarek, 2019-06-17:
Would it be possible to extract relevant parts of firmware and provide the AMBE encoder/decoder as standalone library instead of having to run the full firmware/emulator?

This would also possibly allow to statically reverse-engineer code (or at least provide ASM implementation) and work further on opensource ambe codec implementation

I'm also linking, for relevance, similar idea from GD-77 project https://github.com/rogerclarkmelbourne/Radioddity_GD-77/issues/8

This issue was also written to aggregate relevant info and possibly partial work, that some people may have already done.

rogerclarkmelbourne, 2019-06-17:
Kai DG4KLU has already done this for receive functionality

See https://github.com/talentraspel/GD-77/tree/master/GD-77_new_firmware/firmware

His code is fully open source, except for the AMBE codec binary, which is extracted from the official firmware as 2 separate binary chunks, which are then linked into his firmware.

In the longer term I think it would probably be possible to patch the 2 binaries to make them into one contagious area, and to relocate them to the end of the ROM.

But for the moment we are working with the binaries in their default location from the official firmware.

I did consider doing something similar for the MD-380, because I have one which I have not used for months, but 100% of my available time is currently taken up helping Kai’s with his firmware.

Take a look at my blog www.rogerclark.net for the latest state of play with Kai’s firmware..

And I need to post a new article , because Tx is now partially functional in Kai’s firmware, except that the voice data is empty, because Kai has not had time to do this.
I.e DMR FSK data stream is being produced, albeit Tier1, we  still need to work out how the ADC works in the HR-C6000 and how to run the AMBE codec to encode the data

smarek, 2019-06-20:
I by accident found unpacked Android app with AMBE codec in shared library (.so)
https://github.com/smarek/ambe-codec-p25/

My idea was to extract the md380 AMBE implementation to separate binary/library so it could be used outside of firmware, but i'm now trying to get the ARMv7 compiled shared library to work on different host (or through qemu), but it seems to me that disassemble and re-assemble is the only way, because of Android's non-standard glibc (bionic)

rogerclarkmelbourne, 2019-06-20:
Interesting. I have forked your repo.

Since this is no longer md-380 related it’s probably best discuss this on your new repo, e.g. perhaps o raise a General Chat issue

smarek, 2019-06-20:
Agree, if you do not intend to hold this discussion for extracting md380 firmware ambe implementation to separate library/executable, close it. I'll open general chat in mine repo

DL2MF, 2020-01-06:
Due to the progress of OpenGD77 and lack of relevance for current MD380tools it might be an idea to close this?

travisgoodspeed, 2020-01-07:
Sorry, I didn't see this in June.  The AMBE+2 library from the MD380 can be run under command-line Linux using the instructions on page 38 of `pocorgtfo13.pdf`.  This works by relinking the firmware into a Linux/ARM and then running that under qemu-user.
https://www.alchemistowl.org/pocorgtfo/pocorgtfo13.pdf

Closing issue, as I don't expect to work on the codec emulation anymore.  Any new development will happen elsewhere, likely in OpenGD77.

closed 2020-01-07 by travisgoodspeed