travisgoodspeed/md380tools #904: Support for D014.07 radios

Efpophis, 2019-06-18:
It looks like some changes are (hopefully) in the works to support newer radios that come loaded with D014.07 factory firmware. I have one of these radios, along with several years of embedded development experience, and I'd be glad to help with testing or however I can to get this off the ground.

yuvadm, 2019-07-29:
Adding myself to this thread with the exact same request and offer to help out. AFAIU the main problem with D014.07 firmware is the sleep bug as documented in this thread https://github.com/travisgoodspeed/md380tools/issues/875#issuecomment-453700262

Are there any other problems with the new firmware?

yuvadm, 2019-07-29:
Actually, my previous comment is a bit naive. My current understanding is that the current version of md380tools is based on the D013 firmware, thus being incompatible with the new V5.01 / D014 devices (manifesting in the 5 seconds sleep bug).

Proper D014 support probably means reversing the D014 binaries and adapting the old patches respectively. This might imply a significant effort, but maybe @travisgoodspeed has some tips on how to approach this problem.

dv8awc, 2019-10-29:
there's a fix for the V5.01/D014 md380's to support custom/lower firmware without the rx led and 5s sleep bug @ https://www.facebook.com/groups/KD4ZToolkit

yuvadm, 2019-10-29:
Interesting! Posting the deep link here for posterity https://www.facebook.com/groups/KD4ZToolkit/permalink/707881866388106/

Efpophis, 2019-10-29:
Unfortunately, that version doesn't actually work.  It still has the no analog RX issue, which is a deal-killer.

Also, I sold my MD-380 the other day, so I'm officially out. DMR just isn't my thing.  Good luck.

cyraxx, 2019-10-29:
I had the no analog RX issue on mine but even reflashing back to stock firmware didn't fix it.
Adjusting the squelch calibration values (as shown in the Facebook post) did the trick though. They were way off, I set them to something that seemed more sensible and analog has been fine since.

gibix, 2019-10-30:
There is some chance to have a timeline/todolist for this issue?

travisgoodspeed, 2019-10-31:
Quoting DV8AWC, some form of squelch value editing in the codeplug can fix the issue.

> FIX for the V05.01 MD380's with firnware D014 to use custom/lower firmwares without the rx led and 5s sleep bug. Tested with 2 radios with v5 hardware.
> 
> (no tweaks needed in KD4Z menu)
> 
> You need to change the calibration values for Open SQL 9, Close SQL 9, Open SQL 1, Close SQL 1 via test mode in the CPS. Please see below comments for reference. (NOTE: every radio has its own unique calibration data and is not equally the same)
> 
> Disclaimer: You’ll want to be EXTREMELY careful with these values – do NOT adjust them unless you have proper test equipment! Make sure to backup first your factory test data before making changes.
> 
> -DV8AWC

This can lead to a firmware fix, but we still need to better understand when the firmware should change the settings and what they should be changed to.  (It's not enough to fix the new radios if we break the old ones at the same time.)

--KK4VCZ

yerayad, 2019-12-01:
Here updated to D15.01 to fix the issue with a new radio. 

No RX light and sleep bug until updated with the latest firmware from TYT's website.

chrishelgeson247, 2023-10-17:
Any way to get the md-380 tools to work on a version D015.001?