travisgoodspeed/md380tools #906: UV390 Support

berkin88, 2020-01-04:
Anyone try with Tyt MD-UV390?

travisgoodspeed, 2020-01-05:
Dual band radios are not supported by md380tools.  Sorry.

closed 2020-01-05 by travisgoodspeed

nemanjan00, 2020-12-14:
I fucked up and flashed mine. Any way to fix it?

rogerclarkmelbourne, 2020-12-14:
Just use the official firmware loader to re-flash the official firmware

nemanjan00, 2020-12-14:
I do not have access to windows machine. Only linux. Any other ideas? (or do I need to install windows)

rogerclarkmelbourne, 2020-12-14:
I don't know if any of the tools will upload the official firmware.

You could look at https://github.com/v0l/radio_tool  but I don't know if it supports the UV-380

Probably your best bet would be to find an old Windows computer and use that, or perhaps use a VM.
However the VM will need to support accessing the USB HID device that the radio enumerates as

nemanjan00, 2020-12-14:
Thanks. I did flash it, but, it is locked now

rogerclarkmelbourne, 2020-12-14:
Not sure what you mean

AFIK the bootloader is not changed by the application firmware. 

So its unlikely you completely bricked the radio

The radio will always enter the bootloader if you power it with the buttons on the side pressed in.

You will see it enumerate as a USB HID device (I don't know the VID and PID).

nemanjan00, 2020-12-14:
It is asking for password

rogerclarkmelbourne, 2020-12-14:
What is ?

nemanjan00, 2020-12-14:
Radio turns on now, after flashing original firware, but, it is asking for password

rogerclarkmelbourne, 2020-12-14:
That's because your codeplug has a password set in it.

Try reloading your codeplug again.

BTW. 
I don't know if it is possible to write a codeplug to the radio when its asking for a password.

nemanjan00, 2020-12-14:
`editcp` saved me :D
https://www.farnsworth.org/dale/codeplug/editcp/

rogerclarkmelbourne, 2020-12-14:
OK.