travisgoodspeed/md380tools #909: RX Won't Work Unless Menu Button Pressed

notpike, 2020-04-21:
Hi,

I'm working with a TYT MD-380 with the latest firmware and I'm experiencing something odd and I would like some advice.

While RXing traffic from either a DMR Repeater or Hotspot (NetMon and promiscuous on or off) most radio traffic doesn't get demodulated unless I hit the menu button. It's inconsistent so I'm not sure if this is a sensitivity problem with the radio (some singles are too week to get picked up) or something else like the RX function hanging up.

Any advice would be appreciated.
Thanks!

KD4Z, 2020-04-21:
This might be related to issue #904.   You might give this a try as a possible workaround.   [https://github.com/KD4Z/md380tools-vm/raw/master/Version_5_Hardware_with_MD380tools_firmware.pdf](url)

notpike, 2020-04-23:
Yep I believe that's the same problem I've been experiencing!
May also just have a bum radio as well lol.

Thanks!

closed 2020-04-23 by notpike