travisgoodspeed/md380tools #912: D018 support

plughie, 2020-05-27:
I have an MD-UV380 with D018.003 firmware that the firmware images don't seem to support. My radio doesn't have GPS, but it has a recording feature. The firmware bin is available from the TYT site, and I can send a virgin rdt if it's any help. firmware-NOGPS.bin didn't work (DFU flashing light on reboot).
[md-uv380-recorder-original.zip](https://github.com/travisgoodspeed/md380tools/files/4687358/md-uv380-recorder-original.zip)

mach327, 2020-05-27:
MD-UV380 series - that is, the dual band radios - are not supported, and I don't think anyone has indicated they are currently working on it. 
I'm going to close this ticket, but you can use Github's subscribe feature to keep an eye on #890 in case this changes.

mach327, 2020-05-27:
Added a note in the README just now mentioning this.

closed 2020-05-27 by mach327

rogerclarkmelbourne, 2020-05-28:
FYI. 
I'm going to try to port the OpenGD77 firmware to the Retevis RT3S as soon as my RT3S arrives.

Another group has already started to port the OpenGD77 firmware to the MD-380, AFIK not much is working yet (not even FM).

Luckily the RT3S / UV380 shares the same baseband chip (AT1846S) and DMR chip (HR-C6000) as used in the GD-77, DM-1801 and RD-5R, so the main difference in the port will "only" be the MCU, and the display ;-)

d235j, 2020-05-28:
OpenGD77 is cool, though please note that they are adding claiming it's under the GPL and also "Use of this software for commercial purposes is strictly forbidden" — which is meaningless as the GPL overrides it.

I opened an issue at https://github.com/rogerclarkmelbourne/OpenGD77/issues/604 going into detail as to why this is a problem and it doesn't seem like they are willing to drop that.