travisgoodspeed/md380tools #920: Loses signal after 5 seconds of inactivity. Any button press or channel change will "wake it up"

sunsp0ts, 2021-01-17:
When I purchased this radio a few years ago I had the same problem and returned it. I decided to try it again this holiday season and sure enough I have the exact same problem. I'm a freq with a constant signal.

It goes from RX csbk/idle for a few seconds, to Out_of_sync, to nosig. Once I hit any key on the device it will try again and if there should be traffic this time, it will listen to it. After about 5 seconds everything drops back out. It's incredibly frustrating.

I read on a forum that someone had created a temporary solution where a script would simulate a "dummy button press" every 3 or so seconds to prevent this from happening. I've looked for this supposed script and cannot find it. I then went through the repo here trying to get a handle on the code and figure out a good place and way to write one myself but I'm not sure where to do it. I was hoping someone could help me solve this problem. I know I'm not the only one with this issue. 

I believe travisgoodspeed was sent a radio with this very issue a few years back and he was supposed to look into it but there was never any updates. 

Please help! Otherwise I'll have to sadly return this radio to Amazon for the third time. I have no problem coding it myself with some guidance.

Thanks in advance,

sunsp0ts.

KD4Z, 2021-01-17:
Follow my directions here:
[https://github.com/KD4Z/md380tools-vm/blob/master/Version_5_Hardware_with_MD380tools_firmware.pdf](url)

closed 2021-01-17 by KD4Z

sunsp0ts, 2021-01-17:
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "publisher": { "@type": "Organization", "name": "GitHub", "url":
> Links:

Thank you, these instructions were very difficult to find FYI. I'd 
consider adding links to the threads that complain of issue without a 
resolve. Going to test it now!

73

AC8FM

sunsp0ts, 2021-01-18:
> Closed #920 [1].

Hi Travis,

Unfortunately this did not resolve the issue. I even went so far as to 
increase the values to 32. I'm so frustrated with this radio and the 
amount of times I've spent trying to get it to not go silent. Can't 
there be a simple script to simulate a "wakeup" event that I could 
somehow program into the custom firmware?

Thanks,

Chad AC8FM

sunsp0ts, 2021-01-18:
Hi Again Travis,

So I kept incrementing by 8 until I reached 72! That finally was able to 
hold the squelch open. I'm a little concerned at that high of a number. 
Do you think this could be an issue? I'm still within my Amazon return 
window, I could exchange it for a different radio.

What do you think?

73,

Chad AC8FM

KD4Z, 2021-01-18:
@sunsp0ts   The number is just a calibration value.  It doesn't mean it's being stressed more than a lower value.  Though it may be higher than I have encountered before, it is unique to the hardware that you have. Returning it is your call.

sunsp0ts, 2021-01-18:
> @sunsp0ts [1] The number is just a calibration value. It doesn't mean

Thank you for all of your help. It seems to be working properly. Quick 
question, when scanning, is there anyway to have it display which 
channel/freq it landed on?

Thanks again!