travisgoodspeed/md380tools #921: Trouble Installing on MD-390 (Dual Band, GPS)

dandspach, 2021-02-08:
After spending quite a bit of time trying for myself, I'm stuck...I can't get the MD-380 tools installed.  Any assistance would be welcome.

Radio:  TYT MD-390 ((Dual Band, GPS)
Firmware:  MD-UV390(CSV-GPS)-P18.03
OS:  Windows 10

Trying to install firmware-2021-01-24-GPS.bin.  When using .exe upgrade tool all seems ok, however, when I turn radio off (after flashing, and turn back on, nothing but flashing red and green light.  Have to flash MD-UV390(CSV-GPS)-P18.03 back to radio.  

Is there some limitation to the MD-380Tools software that can be installed?

dandspach, 2021-02-08:
Maybe I just found the answer to my question...perhaps its not compatible with UV version of this radio...

rogerclarkmelbourne, 2021-02-08:
See the ReadMe.md

_Dual band radios such as the MD2017 and MD-UV380 series are not supported._

closed 2021-02-09 by dandspach