travisgoodspeed/md380tools #922: Fail to make flash on Arch Linux multiple definition

toddejohnson, 2021-02-13:
I'm trying to update my MD380 I hadn't updated since 2017(03a05ffbcfd3e016db78f5b550cd21d50ccc840e). Been working great! Thank you for that. I pulled to current (1453fe37d06b5f428d7919fd0670e7e427de9457) and installed the arm stuff. Possibly on a new laptop running Arch Linux. I know it isn't listed with build prep instructions but I installed the needed files:

```sh
pacman -S --needed arm-none-eabi-binutils arm-none-eabi-gcc arm-none-eabi-newlib libusb python-pyusb python-requests make curl
```

This installed:
```
arm-none-eabi-binutils 2.35.1-1
arm-none-eabi-gcc 10.2.0-1
arm-none-eabi-newlib 4.1.0-1
libusb 1.0.24-2
python-pyusb 1.1.0-3
python-requests 2.25.1-1
make 4.3-3
curl 7.75.0-1
```

I think there is a change in gcc as I'm seeing a bunch of multiple definition errors:
```
arm-none-eabi-gcc -std=gnu99 -g -O2 -Wall -Tapplet_d13.020.ld -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fshort-wchar -Wa,-a=applet.elf -DFW_D13_020   -Iinc -Ilib -Ilib/inc -I. -Ilib/inc/core -Ilib/inc/peripherals -D "MD380_d13_020"   -Wl,--no-wchar-size-warning -Wl,-Map,applet.map main.o printf.o dmesg.o gfx.o system_stm32f4xx.o addl_config.o spiflash.o usb.o menu.o rtc_timer.o usersdb.o os.o aes.o ambe.o dmr.o display.o console.o util.o netmon.o debug.o keyb.o radiostate.o mbox.o syslog.o lastheard.o etsi.o sms.o beep.o irq_handlers.o system_hrc5000.o narrator.o lcd_driver.o font_8_8.o app_menu.o color_picker.o amenu_hexmon.o amenu_utils.o amenu_set_tg.o amenu_channels.o amenu_codeplug.o timer.o dispinit.o lib/startup_stm32f4xx_asm.S -o applet.elf -Llib -lstm32f4 -Xlinker --just-symbols=src/symbols_d13.020
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: gfx.o:/home/todd/git/md380tools/applet/src/md380.h:187: multiple definition of `gui_opmode1'; main.o:/home/todd/git/md380tools/applet/src/md380.h:187: first defined here
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: gfx.o:/home/todd/git/md380tools/applet/src/md380.h:124: multiple definition of `md380_menu_id'; main.o:/home/todd/git/md380tools/applet/src/md380.h:124: first defined here
```
I managed to fix some of these by adding `extern uint8_t md380_menu`... to many of the errors however this doesn't fix kc_ in keyb.h. This yields undefined reference to errors which are a bit outside of my skill in Google and C. Full output is attached [buildlog.txt](https://github.com/travisgoodspeed/md380tools/files/5975166/buildlog.txt) and with my extern changes [buildlog2.txt](https://github.com/travisgoodspeed/md380tools/files/5975165/buildlog2.txt). 

I can happily open a PR with the changes I have for extern but that doesn't fix it. Any ideas what I need to do to get the variable defined in this "newer?" version of gcc?

halfmanhalftaco, 2021-05-29:
Adding `-fcommon` to CFLAGS in applet/Makefile fixes build issues on arm-none-eabi-gcc 11.1.0 on Arch Linux for me.

travisgoodspeed, 2021-05-29:
I'll spin up an Arch Linux VM, to have a more modern toolchain.  In the meantime, could one of you make a PR with the CFLAGS change?

toddejohnson, 2021-05-29:
Wow @halfmanhalftaco that got it down to a one line fix. Wish I knew C like that! I dumped a PR but @halfmanhalftaco should get the credit.

ast, 2023-02-12:
I had exactly the same problem on Ubuntu 22.04.

Adding

`CFLAGS += -fcommon`

After line 93 in `applet/Makefile` solved it.

I also had to install python2, as it's not installed by default anymore.

leonidbelyaev, 2023-05-06:
The -fcommon flag doesn't fix the issue for me in NixOS.

cfenell, 2025-08-03:
I managed to upgrade my MD-380 today using an Arch Linux laptop. Here is what I had to do

1. check out Python3 branch
2.  add to `applet/Makefile`:  CFLAGS += -fcommon -specs=nosys.specs
3. download the firmware listed in `firmware_files.txt` by hand, the grep commands fail somehow
3. unwrap downloaded firmware into firmware/unwrapped/D013.020.img

After this the build succeeds.

*md380-dfu* fails since it cannot interpret a Request object as int.
So I used the online tool [](https://dmr.tools) to load and flash `applet/experiment.bin`

Hope this helps
Fredrik SM2YHP