travisgoodspeed/md380tools #925: md380-emu crashes on all recent arm platfrom OS versions

RadioRedFred, 2021-04-11:
For some months now, the md380-emu example, as well as all of the ham radio cross mode tools out there using md380 firmware for transcoding, using D02.032 firmware crashes on all of my platforms that used to work.  This includes pistar on rpi 2,3,4 and raspiOS on same pi's.  It all started with updates to these operating systems, but I don't know exactly when.  As it stands, pi-star 4.1.4 with latest updates crashes, and raspiOS with latest updates also crashes.

RaspiOS fresh install from the 01-11-2021 image *does not crash*.  md380-emu decode example (./md380-emu -d -i sample.amb -o out.raw) as well as the encode example work correctly on pi2,3,4 using this OS as long as I do not do any updates.

nostar, 2021-09-22:
I submitted a fix for this here https://github.com/travisgoodspeed/md380tools/pull/932

travisgoodspeed, 2021-09-23:
Thanks for sending in a fix.  I'll give it a try in the morning and merge it if everything looks good.

RdWing, 2024-08-07:
> Thanks for sending in a fix. I'll give it a try in the morning and merge it if everything looks good.

Is this the cause of 
`<user>@raspberrypi:~ $ /usr/bin/qemu-arm-static /opt/md380-emu/md380-emu -S 2470
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault`
If so, can we get this merged in?

closed 2024-10-26 by travisgoodspeed