travisgoodspeed/md380tools #926: Radio Stops listening after 5 seconds of inactivity

joshs85, 2021-05-09:
Hi, on certain versions of the MD380 Non GPS hardware, once flashed and patched with md380 tools, the radio will not work properly.   When you first turn it on, it will listen for anything on whatever chanel its on for about 5 seconds.  If theres a transmission after that, the radio won't make any noise.  If I click the menu button while there is a transmission going over the air, it will all of the sudden start playing.. then it will work until there's another 5 seconds of dead air.  This was fixed on my radio previously by implimenting the nosleep.enable fix but it doesn't seem to be working on this version.  Could you re add this please?  I found a reference to it here.. but couldn't find any more information anywhere.  

https://github.com/KD4Z/md380tools-vm/blob/master/init#L98-L106

KD4Z, 2021-05-10:
See
https://github.com/KD4Z/md380tools-vm/blob/master/Version_5_Hardware_with_MD380tools_firmware.pdf

joshs85, 2021-05-10:
That fixed it.  Thanks!

closed 2021-05-10 by joshs85