travisgoodspeed/md380tools #954: reboot loop when using 7-digit TG

RadioMan57, 2024-07-15:
(1) md380tools 2.0 09e0f4a goes into reboot loop

(2) a) use 7-digit TG (e.g. 8002280) as required when operating via a hotspot
(2) b) Mode Display = Mode compact, or Channel Info = Ch Status Info
(2) c) select channel with knob

(3) truncate long TGs if they don't fit in the display

Markus HB9BRJ

RadioMan57, 2024-10-16:
Update 2014-10-16:
TG 500xxxx is accepted without any problem, TG 800xxxx freezes the MD-380.
Is there a maximum TG value?

travisgoodspeed, 2024-10-16:
16777215 should be  the largest talkgroup number possible.

Does this same problem exist in the unpatched manufacturer firmware?

RadioMan57, 2024-10-26:
Did some more experiments:
Any incoming TG higher than the maximum DMR ID in the user database causes the MD380 to freeze or reboot. After erasing the user database, every TG up to the "legal maximum" 16777215 is accepted.

closed 2024-10-26 by RadioMan57