travisgoodspeed/md380tools #958: make dist fails on macOS 15.3

msmuenchen, 2025-02-13:
Hi all,

for current git state, on macOS compilation fails with a bunch of warnings.

Relevant versions:

```
arm-none-eabi-gcc              @14.2.0         (Macports)
Python 3.12.8 (via python-virtualenv)
pip 24.3.1 from /Users/xxxxxx/Projects/md380tools/md380-venv/lib/python3.12/site-packages/pip (python 3.12)
```

Machine is a 2022 MBA with M2 CPU.

Log is attached as file.

[log.txt](https://github.com/user-attachments/files/18783010/log.txt)

73 de DN9AFA

mikeazo, 2025-02-13:
I created a [fork](https://github.com/mikeazo/md380tools) a while back to build this repo on an Intel mac. Basically had to change some compiler flags and some of the USB interfacing code (see [commit](https://github.com/mikeazo/md380tools/commit/97d6e20e14c7d1d121ff61dfd841cc9bb56e7eb0)).

I haven't tried this on Apple Silicon though. My repo is a few commits behind. If my fixes make it so you can build, I can take a look at the commits since and make sure they won't break anything.

msmuenchen, 2025-02-13:
@mikeazo I'll have to check that one today evening and will report back, thanks.

What I immediately see is you're hardcoding the libusb path to "/opt/local/lib/libusb.dylib", while macports libusb is "/opt/local/lib/libusb-1.0.dylib". Your path is only valid if you install libusb-compat as well, and in any case it's only valid if you use macports. Homebrew will have completely different paths but I refuse to touch this with a ten foot pole...

mikeazo, 2025-02-13:
@msmuenchen thanks for the analysis. That is exactly why I never submitted a pull request, I figured it wasn't a universal fix. It worked for me, but I need to do more testing (on my M3 mac). 

I expect the compiler flags will fix your build errors. The python changes were runtime fixes, and it sounds like you know what needs to be done there to make it work.

msmuenchen, 2025-02-13:
Unfortunately I only barely speak Python, I'm a sysop by trade so that one was immediately obvious ;) I'll do my best to check up.